repaper/gratis #20: Old display works, but not a new one.

radiolok, 2015-02-13:
Hi!
I use pervasive display boosterpacks with 2 inch EPD and EPD library. And use MSP430F5529 launchpads. 
I broke one display and hed to buy new one.
All what happend with new display - after first attempt to show image there is only some grey lines on it. After 3-4 tries whole display is black and nothing happens.
When I replace this display with an old one - it works good.

You may think, that the problem is in the display module, but I have 4 new displays with the same behaviour!.
Sometime, after turning on, display became white, but blacky behaviour repited.

If you show on the serial number, you can see, that driver revision(Symbol 7 = 'A') is the same, but new displays serial number starts from 'R' symbol - I didn't find anything about this.
Old display serial:
NEG020AS0122A90212
New display:
REG020AS0123B82031

asierazaceta, 2015-02-13:
Hi,
When did you purchased each of the packs? Because during last year PDI discontinued some models and were replaced by new part numbers with different COG's and thus drivers.
According to the serial numbers you send that shouldn't be the case, but it may be worth checking it.
I haven't worked with boosterpacks for EPDs but rather with the EPDs directly so I don't know if you are referring to the pack serial number or the EPD serial number.
For your reference, as far as I know, the part number from those serial numbers would be:
EG020AS012
Which looks the same in both cases. Unfortunately that's the OLD part number, I mean, the part number for the discontinued EPD, which has been replaced by the
EG020BS011

Again, I may be wrong, but those were my 2 cents.
Hope it helps.

radiolok, 2015-02-13:
I bought old display in May'14 and new one at Sept'14.

radiolok, 2015-02-16:
I found and solve a problem - MSP430F5529 launchpads don't have PWM signal on pin 9, so I have to redefine PWM to pin 30 and made a jumper between this two pins.

During datasheet you need PWM with revision A display and do not need with new revision B.

Notice, that in Boosterpackpinout pin 29 and 30 marked as reserved.

closed 2015-02-16 by radiolok