repaper/gratis #21: Rotating Screen

danielbk18, 2015-03-27:
Hi, I'm trying to use the screen rotated, is there a way to update the columns serially instead of lines? The way described of sending the line and then the scan bytes, is it the only way to update the screen? For example, I would like to send just one byte in a specific column and line to make a faster partial update, without sending 'nothing' bytes to the rest of the screen.

CharmingSu, 2015-03-30:
Hi,
The line data and scan bytes order are fixed due to the driver buffer and traces of TFT. If you want to rotate the content, you have to convert the source image to the orientation beforehand.
Your another request is the [Partial Update](http://www.pervasivedisplays.com/kits/ext_kit#tab6) driving. See [my comments at #19](https://github.com/repaper/gratis/issues/19#issuecomment-68039046).
By FAE Charming Su of Pervasive Displays

CharmingSu, 2015-03-30:
New V231 display with G2 driver are available in March 2015 on Digi-Key!

- 1.44": [E1144CS021](http://www.digikey.com/product-detail/en/E1144CS021/E1144CS021-ND/5046789)
- 1.9" : [E1190CS021](http://www.digikey.com/product-detail/en/E1190CS021/E1190CS021-ND/5046790)
- 2.0" : [E1200CS021](http://www.digikey.com/product-detail/en/E1200CS021/E1200CS021-ND/5046791)
- 2.6" : [E1260CS021](http://www.digikey.com/product-detail/en/E1260CS021/E1260CS021-ND/5046792)
- 2.7" : [E1270CS021](http://www.digikey.com/product-detail/en/E1270CS021/E1270CS021-ND/5046793)

The Gratis Github code is upgrading.

For more information of the difference, please visit [Pervasive Displays](http://www.pervasivedisplays.com/products/panels#tab2).
How to verify your display type, please check the rear label of your display by [this link](http://www.pervasivedisplays.com/products/label_info). 

By FAE - Charming Su of Pervasive Displays

closed 2015-06-01 by hxw