repaper/gratis #23: writte error on Beaglebone Black

doganyazar, 2015-05-01:
I am getting a write error on Beaglebone black with Debian. 
EPD-fuse service seems to work fine since I can access the panel/version etc. However, whenever I want to send a command, first I get a connection abort and then endpoint is not connected. 

Any ideas about what may have gone wrong? Is this ever tested on Debian? 

```
~/gratis/PlatformWithOS$ sudo service epd-fuse restart

~/gratis/PlatformWithOS$ cat /dev/epd/panel                                                                        
EPD 2.0 200x96 COG 1 FILM 110
                                           
~/gratis/PlatformWithOS$ python demo/DrawDemo.py                                                                             panel = EPD 2.0 200 x 96  version=4 COG=1 FILM=110
Traceback (most recent call last):
  File "demo/DrawDemo.py", line 70, in <module>
    main(sys.argv[1:])
  File "demo/DrawDemo.py", line 31, in main
    epd.clear()
  File "/home/scypho/gratis/PlatformWithOS/demo/EPD.py", line 152, in clear
    self._command('C')
  File "/home/scypho/gratis/PlatformWithOS/demo/EPD.py", line 156, in _command
    f.write(c)
IOError: [Errno 103] Software caused connection abort

~/gratis/PlatformWithOS$ cat /dev/epd/panel                                                                                  
cat: /dev/epd/panel: Transport endpoint is not connected

~/gratis/PlatformWithOS$ sudo service epd-fuse restart

~/gratis/PlatformWithOS$ echo -n "C" > /dev/epd/command                                                                      
-bash: echo: write error: Software caused connection abort

~/gratis/PlatformWithOS$ echo -n "C" > /dev/epd/command
-bash: /dev/epd/command: Transport endpoint is not connected

```

hxw, 2015-05-07:
Debian has not been tested in a while.  Currenlty I run Arch.
Can you see if the right SPI driver is loaded, possible you have to edit a config file in the boot(FAT) partition to enable this.  Would you post the Debian version you are using:  and the "uname -a" output.

hxw, 2015-05-07:
I was able to get it working with:
http://debian.beagleboard.org/images/bone-debian-7.8-lxde-4gb-armhf-2015-03-01-4gb.img.xz

see som bief instructions at: https://github.com/repaper/gratis/blob/master/PlatformWithOS/README-BBB-Debian.md

doganyazar, 2015-05-07:
Thank you very much for your answer. 
After I asked this question I continued digging. I tried them with Debian kernel 3.14 and 3.17 and both failed and my guess is that those images does not have the SPI support. The solutions I have found suggest enabling SPI via capemgr but it does not exist in images with kernel > 3.8 according to the following link:
https://eewiki.net/display/linuxonarm/BeagleBone+Black+Comments?focusedCommentId=34537554#comment-34537554
“The beaglebone capemgr only still works 100% in 3.8.x, i removed it from unsupported/unworking versions, while we wait for kernel.org developers to agree on the "overlay" patchset.”

The image you have tried apparently has 3.8 kernel so I guess it makes sense that it works. But I need a newer kernel for my project. 
I know I am asking too much but Is it possible that you can try with a newer kernel as well? 

One last thing, apparently the firmware you guys using for the GPIO pins (https://github.com/nomel/beaglebone.git) is deprecated now and the github repository suggests using
https://github.com/cdsteinkuehler/beaglebone-universal-io

hxw, 2015-05-08:
Is there a specific kernel version should try?
or are you building your own kernel on an a particular distribution?
(Could i build on the 2015-03-01 install I made yesterday?)

The Arch system (currently running) shows that there appears to be a 3.8.18 kernel available
I could try this *assuming 3.18 is a high enough version*
~~~~~
% pacman -Ss linux-am33                                                                                                                                    
core/linux-am33x 3.18.2-1
    The Linux Kernel and modules - TI AM335x Beaglebone (Black)
core/linux-am33x-headers 3.18.2-1
    Header files and scripts for building modules for linux kernel - TI AM335x Beaglebone (Black)
core/linux-am33x-legacy 3.8.13-37 [installed]
    The Linux Kernel and modules - am33x processors
~~~~~

On the universal IO after a brief look it seems all gpios are in one dts file now so the that file needs to be checked first in BeagleBone/gpio.c   probably need to only compile/install **cape-universal-00A0.dts** - I will also look at this; thanks for pointing out the change.

doganyazar, 2015-05-08:
3.18 is definitely high enough :) 
3.14 version I tried is from the beaglebone black debian jessie image  
http://elinux.org/Beagleboard:BeagleBoneBlack_Debian#Jessie_Snapshot_console

Thanks :)

hxw, 2015-05-15:
*Lack of* Progress Report

It looks like later kernels do not contain the cape manager and I do not think that support will be added.   I found very few asking for such support.
Perhaps this is due to Linus refusal to accept any more ARM SOC specific files .
I can really understand how all these SOCs with their seemingly random combination of peripherals and I/O multiplexing would make maintaining suc files impossible. 
The approved solution is to use device tree files;  I see that there are several different dtb files in 
`/boot/dtbs`.  On linux 4.0 I found several different bone/boneblack files one has "cape" in its name.
the dtc programs is able to show source from these binaries.  The cape-universal souce does not appear to be compilable by the latest dtc.

I think the next step is for me to find out how to load a different dtb file via u-boot (need to fid out how the uEnv.txt in the Fat partition does this).  The next step is to make a "repaper" dts file by combining the boneblack and universal files and naming the pins like "EPD-RESET"; hopefully the resulting file is usable across all distros (maybe even portable to RPi...).

I still need to research the "dts" syntax to be able to do this, there is a lot to read, so progress is slow.

doganyazar, 2015-06-24:
I wanted to spend time on this and figure out exactly how things are but I am quite lost :) Did not quite understand what "The cape-universal source does not appear to be compilable by the latest dtc." means either. 
What I understood is if we use one of the ready images done by Robert Nelson (RCN), then it should be enough to enable universal device tree overlay and configure SPI via config-pin (if necessary). So there should not be any need to compile any dts files and boot with the generated dtb files. Please correct me if I am missing something. 
(https://github.com/cdsteinkuehler/beaglebone-universal-io, make && make install)
Anyway I will dig into this a bit more and try out different options. 

Besides, according to this (https://github.com/beagleboard/linux/issues/27) kernel 4.1 has cape support.

doganyazar, 2015-06-25:
And this fresh discussion tells how to use capemgr with kernel 4.1:
http://beagleboard.org/Community/Forums?place=msg%2Fbeagleboard%2FviIKOGegUko%2FsGC4r3Xl3dMJ

doganyazar, 2015-07-06:
Some update on this, 
I have tried "Flasher: (lxqt-4gb) (BeagleBone Black 4GB eMMC)" image. It comes with kernel 4.1 and beaglebone-universal-io (https://github.com/cdsteinkuehler/beaglebone-universal-io)
http://elinux.org/Beagleboard:BeagleBoneBlack_Debian#Jessie_Snapshot_lxqt

debian@dodo4:~$ config-pin overlay cape-universal 
Loading cape-universal overlay
debian@dodo4:~$ ls /dev/spidev*
/dev/spidev1.0  /dev/spidev1.1  /dev/spidev2.0  /dev/spidev2.1

All good now I have spi as well. But still epaper does not work since gpio configuration does not seem to work. 
~/gratis/PlatformWithOS$ driver-common/gpio_test 
gpio_test: GPIO_setup failed: Success

I could not find enough time to figure out GPIO pin settings but I assume with the latest beaglebone image, the way to set GPIO is to use 
config-pin command line tool (from universal-io). Sth like config-pin P8.07 hi

Any suggestions on what I can try to make gpio settings work?

hxw, 2015-07-06:
are you able to determine the pin names? 
in BeagleBone/gpio.c  I created them as:
``` 
        MAKE_PIN("gpio-P8.03", 1, 6),   //  GPIO1_6
```
and if the names hae just changed then altering the "gpio.PX.YY" strings to match might work

doganyazar, 2015-07-06:
I am not exactly sure but the folder structure described in your code comments do not seem to be valid anymore (they do not exist!)
const char *name;   // e.g. "gpio-P8.15" -> /lib/firmware/gpio-P8.15.dtbo
char *state;        // e.g. "/sys/devices/ocp.3/gpio-P8.15_gpio47.17/state"

Instead:
~/gratis/PlatformWithOS$ ls -al /sys/devices/platform/ocp/ | grep ocp
drwxr-xr-x  3 root root    0 Jul  6 12:28 ocp:P8_07_pinmux
drwxr-xr-x  3 root root    0 Jul  6 12:28 ocp:P8_08_pinmux
drwxr-xr-x  3 root root    0 Jul  6 12:28 ocp:P8_09_pinmux
drwxr-xr-x  3 root root    0 Jul  6 12:28 ocp:P8_10_pinmux
drwxr-xr-x  3 root root    0 Jul  6 12:28 ocp:P8_11_pinmux
drwxr-xr-x  3 root root    0 Jul  6 12:28 ocp:P8_12_pinmux
drwxr-xr-x  3 root root    0 Jul  6 12:28 ocp:P8_13_pinmux
drwxr-xr-x  3 root root    0 Jul  6 12:28 ocp:P8_14_pinmux
drwxr-xr-x  3 root root    0 Jul  6 12:28 ocp:P8_15_pinmux
drwxr-xr-x  3 root root    0 Jul  6 12:28 ocp:P8_16_pinmux
drwxr-xr-x  3 root root    0 Jul  6 12:28 ocp:P8_17_pinmux
drwxr-xr-x  3 root root    0 Jul  6 12:28 ocp:P8_18_pinmux
drwxr-xr-x  3 root root    0 Jul  6 12:28 ocp:P8_19_pinmux
drwxr-xr-x  3 root root    0 Jul  6 12:28 ocp:P8_26_pinmux
drwxr-xr-x  3 root root    0 Jul  6 12:28 ocp:P9_11_pinmux
drwxr-xr-x  3 root root    0 Jul  6 12:28 ocp:P9_12_pinmux
drwxr-xr-x  3 root root    0 Jul  6 12:28 ocp:P9_13_pinmux
drwxr-xr-x  3 root root    0 Jul  6 12:28 ocp:P9_14_pinmux
drwxr-xr-x  3 root root    0 Jul  6 12:28 ocp:P9_15_pinmux
drwxr-xr-x  3 root root    0 Jul  6 12:28 ocp:P9_16_pinmux
drwxr-xr-x  3 root root    0 Jul  6 12:28 ocp:P9_17_pinmux
drwxr-xr-x  3 root root    0 Jul  6 12:28 ocp:P9_18_pinmux
drwxr-xr-x  3 root root    0 Jul  6 12:28 ocp:P9_21_pinmux
drwxr-xr-x  3 root root    0 Jul  6 12:28 ocp:P9_22_pinmux
drwxr-xr-x  3 root root    0 Jul  6 12:28 ocp:P9_23_pinmux
drwxr-xr-x  3 root root    0 Jul  6 12:28 ocp:P9_24_pinmux
drwxr-xr-x  3 root root    0 Jul  6 12:28 ocp:P9_25_pinmux
drwxr-xr-x  3 root root    0 Jul  6 12:28 ocp:P9_26_pinmux
drwxr-xr-x  3 root root    0 Jul  6 12:28 ocp:P9_27_pinmux
drwxr-xr-x  3 root root    0 Jul  6 12:28 ocp:P9_28_pinmux
drwxr-xr-x  3 root root    0 Jul  6 12:28 ocp:P9_29_pinmux
drwxr-xr-x  3 root root    0 Jul  6 12:28 ocp:P9_30_pinmux
drwxr-xr-x  3 root root    0 Jul  6 12:28 ocp:P9_31_pinmux
drwxr-xr-x  3 root root    0 Jul  6 12:28 ocp:P9_42_pinmux
drwxr-xr-x  3 root root    0 Jul  6 12:28 ocp:P9_91_pinmux
drwxr-xr-x  3 root root    0 Jul  6 12:28 ocp:P9_92_pinmux
drwxr-xr-x  3 root root    0 Jul  6 12:28 ocp:cape-universal
drwxr-xr-x  5 root root    0 Jul  6 12:28 ocp:l4_wkup@44c00000
lrwxrwxrwx  1 root root    0 Jul  6 12:28 of_node -> ../../../firmware/devicetree/base/ocp

debian@dodo4:~$ ls -al /lib/firmware/
total 4724
drwxr-xr-x 11 root root   12288 Jun 29 07:13 .
drwxr-xr-x 17 root root    4096 Jun 29 07:00 ..
-rw-r--r--  1 root root    1042 Jun 29 07:12 ADAFRUIT-SPI0-00A0.dtbo
-rw-r--r--  1 root root    1115 Jun 29 07:12 ADAFRUIT-SPI1-00A0.dtbo
-rw-r--r--  1 root root     865 Jun 29 07:12 ADAFRUIT-UART1-00A0.dtbo
-rw-r--r--  1 root root     865 Jun 29 07:12 ADAFRUIT-UART2-00A0.dtbo
-rw-r--r--  1 root root     865 Jun 29 07:12 ADAFRUIT-UART4-00A0.dtbo
-rw-r--r--  1 root root     865 Jun 29 07:12 ADAFRUIT-UART5-00A0.dtbo
-rw-r--r--  1 root root     510 Jun 29 07:13 BB-ADC-00A0.dtbo
-rw-r--r--  1 root root    2398 Jun 29 07:13 BB-BBBMINI-00A0.dtbo
-rw-r--r--  1 root root    2310 Jun 29 07:13 BB-BONE-AUDI-02-00A0.dtbo
-rw-r--r--  1 root root    2964 Jun 29 07:13 BB-BONE-BACONE-00A0.dtbo
-rw-r--r--  1 root root    1335 Jun 29 07:13 BB-BONE-CRYPTO-00A0.dtbo
-rw-r--r--  1 root root    2247 Jun 29 07:13 BB-BONE-DVID-01-00A3.dtbo
-rw-r--r--  1 root root    5212 Jun 29 07:13 BB-BONE-LCD3-01-00A2.dtbo
-rw-r--r--  1 root root    5209 Jun 29 07:13 BB-BONE-LCD4-01-00A1.dtbo
-rw-r--r--  1 root root    5209 Jun 29 07:13 BB-BONE-LCD7-01-00A3.dtbo
-rw-r--r--  1 root root    6267 Jun 29 07:13 BB-BONE-REPLICAP-00B1.dtbo
-rw-r--r--  1 root root     867 Jun 29 07:13 BB-BONE-SERL-03-00A1.dtbo
-rw-r--r--  1 root root     851 Jun 29 07:13 BB-CAN1-00A0.dtbo
-rw-r--r--  1 root root     940 Jun 29 07:13 BB-I2C1-00A0.dtbo
-rw-r--r--  1 root root    1901 Jun 29 07:13 BB-I2C1-PCA9685-00A0.dtbo
[... 141 lines cut ...]
17 P9_15                     48 IN  0
18 P9_14                     50 IN  0
19 P9_13                     31 IN  0
20 P9_12                     60 IN  0
21 P9_11                     30 IN  0
22 P8_26                     61 IN  0
23 P8_19                     22 IN  0
24 P8_18                     65 IN  0
25 P8_17                     27 IN  0
26 P8_16                     46 IN  0
27 P8_15                     47 IN  0
28 P8_14                     26 IN  0
29 P8_13                     23 IN  0
30 P8_12                     44 IN  0
31 P8_11                     45 IN  0
32 P8_10                     68 IN  0
33 P8_09                     69 IN  0
34 P8_08                     67 IN  0
35 P8_07                     66 IN  0

And hence I can reach the pins
~/gratis/PlatformWithOS$ cat /sys/class/gpio/gpio47/direction                                                                
in
~/gratis/PlatformWithOS$ cat /sys/class/gpio/gpio47/value
0

hxw, 2015-07-09:
After initian boot are all the pins already set up?
i.e. does `ls -l /sys/class/gpio/` sow all the GPIO already active or do they only apear after `config-pin overlay cape-universal` has been run.

Not sure if you can attach LED to a GPIO and echo OUT into its direction then echo 0 or 1 to its value and see if it can light.

hxw, 2015-07-09:
I just updated my BBB arch to:
`Linux alarm 4.1.1-2-ARCH #1 Mon Jul 6 21:13:46 MDT 2015 armv7l GNU/Linux`
and I enable gpio by:
~~~
# echo 47 >/sys/class/gpio/export 
[root@alarm ~]# ls -l /sys/class/gpio/                                                                                                                                              
total 0
--w------- 1 root root 4096 Jul  9 11:05 export
lrwxrwxrwx 1 root root    0 Jul  9 11:05 gpio47 -> ../../devices/platform/ocp/4804c000.gpio/gpio/gpio47
lrwxrwxrwx 1 root root    0 Jul  9 11:00 gpiochip0 -> ../../devices/platform/ocp/44e07000.gpio/gpio/gpiochip0
lrwxrwxrwx 1 root root    0 Jul  9 11:00 gpiochip32 -> ../../devices/platform/ocp/4804c000.gpio/gpio/gpiochip32
lrwxrwxrwx 1 root root    0 Jul  9 11:00 gpiochip64 -> ../../devices/platform/ocp/481ac000.gpio/gpio/gpiochip64
lrwxrwxrwx 1 root root    0 Jul  9 11:00 gpiochip96 -> ../../devices/platform/ocp/481ae000.gpio/gpio/gpiochip96
--w------- 1 root root 4096 Jul  9 11:05 unexport

[root@alarm ~]# cat /sys/class/gpio/gpio47/direction                                                                                                                                
in
[root@alarm ~]# echo out > /sys/class/gpio/gpio47/direction                                                                                                                         
[root@alarm ~]# cat /sys/class/gpio/gpio47/direction                                                                                                                                
out
~~~
now to get LED and see if I can flash it

doganyazar, 2015-07-12:
Before I enable any cape:
$ ls -al /sys/class/gpio/
total 0
drwxr-xr-x  2 root root    0 Jul 12 20:44 .
drwxr-xr-x 57 root root    0 Jul 12 20:44 ..
--w-------  1 root root 4096 Jul 12 20:44 export
lrwxrwxrwx  1 root root    0 Jul 12 20:44 gpiochip0 -> ../../devices/platform/ocp/44e07000.gpio/gpio/gpiochip0
lrwxrwxrwx  1 root root    0 Jul 12 20:44 gpiochip32 -> ../../devices/platform/ocp/4804c000.gpio/gpio/gpiochip32
lrwxrwxrwx  1 root root    0 Jul 12 20:44 gpiochip64 -> ../../devices/platform/ocp/481ac000.gpio/gpio/gpiochip64
lrwxrwxrwx  1 root root    0 Jul 12 20:44 gpiochip96 -> ../../devices/platform/ocp/481ae000.gpio/gpio/gpiochip96
--w-------  1 root root 4096 Jul 12 20:44 unexport

I think the config-pin command is just a wrapper for the manual export and set operations you did. It is just the command line tool of the  beaglebone-universal-io github library and it is installed in the beaglebone images: http://elinux.org/Beagleboard:BeagleBoneBlack_Debian#Jessie_Snapshot_lxqt

What I understood is that enabling the universal cape just exports all the pins and enable the devices. 
Then you could just write "config-pin P9.12 hi" to set it. But actually it did not really export anything under /sys/class/gpio/ and it does not allow me to set it either so I guess something is wrong :) 

I will check this further tomorrow.

doganyazar, 2015-07-13:
I tried it again and universal cape seems to work fine. I did the same thing you did (connect a led) and turned it on and off doing the following:
$ config-pin overlay cape-universal
$ config-pin P8.10 out
$ config-pin P8.10 hi
$ config-pin P8.10 low

After I run "$ config-pin overlay cape-universal" , the pins are exported successfully as well. 
debian@dodo4:~$ ls -al /sys/class/gpio/
total 0
drwxr-xr-x  2 root root    0 Jul 13 09:24 .
drwxr-xr-x 57 root root    0 Jul 13 09:24 ..
--w-------  1 root root 4096 Jul 13 09:23 export
lrwxrwxrwx  1 root root    0 Jul 13 09:24 gpio110 -> ../../devices/platform/ocp/481ae000.gpio/gpio/gpio110
lrwxrwxrwx  1 root root    0 Jul 13 09:24 gpio111 -> ../../devices/platform/ocp/481ae000.gpio/gpio/gpio111
lrwxrwxrwx  1 root root    0 Jul 13 09:24 gpio112 -> ../../devices/platform/ocp/481ae000.gpio/gpio/gpio112
lrwxrwxrwx  1 root root    0 Jul 13 09:24 gpio113 -> ../../devices/platform/ocp/481ae000.gpio/gpio/gpio113
lrwxrwxrwx  1 root root    0 Jul 13 09:24 gpio114 -> ../../devices/platform/ocp/481ae000.gpio/gpio/gpio114
lrwxrwxrwx  1 root root    0 Jul 13 09:24 gpio115 -> ../../devices/platform/ocp/481ae000.gpio/gpio/gpio115
lrwxrwxrwx  1 root root    0 Jul 13 09:24 gpio116 -> ../../devices/platform/ocp/481ae000.gpio/gpio/gpio116
lrwxrwxrwx  1 root root    0 Jul 13 09:24 gpio117 -> ../../devices/platform/ocp/481ae000.gpio/gpio/gpio117
lrwxrwxrwx  1 root root    0 Jul 13 09:24 gpio14 -> ../../devices/platform/ocp/44e07000.gpio/gpio/gpio14
lrwxrwxrwx  1 root root    0 Jul 13 09:24 gpio15 -> ../../devices/platform/ocp/44e07000.gpio/gpio/gpio15
lrwxrwxrwx  1 root root    0 Jul 13 09:24 gpio2 -> ../../devices/platform/ocp/44e07000.gpio/gpio/gpio2
lrwxrwxrwx  1 root root    0 Jul 13 09:24 gpio22 -> ../../devices/platform/ocp/44e07000.gpio/gpio/gpio22
lrwxrwxrwx  1 root root    0 Jul 13 09:24 gpio23 -> ../../devices/platform/ocp/44e07000.gpio/gpio/gpio23
lrwxrwxrwx  1 root root    0 Jul 13 09:24 gpio26 -> ../../devices/platform/ocp/44e07000.gpio/gpio/gpio26
lrwxrwxrwx  1 root root    0 Jul 13 09:24 gpio27 -> ../../devices/platform/ocp/44e07000.gpio/gpio/gpio27
lrwxrwxrwx  1 root root    0 Jul 13 09:24 gpio3 -> ../../devices/platform/ocp/44e07000.gpio/gpio/gpio3
lrwxrwxrwx  1 root root    0 Jul 13 09:24 gpio30 -> ../../devices/platform/ocp/44e07000.gpio/gpio/gpio30
lrwxrwxrwx  1 root root    0 Jul 13 09:24 gpio31 -> ../../devices/platform/ocp/44e07000.gpio/gpio/gpio31
lrwxrwxrwx  1 root root    0 Jul 13 09:24 gpio4 -> ../../devices/platform/ocp/44e07000.gpio/gpio/gpio4
lrwxrwxrwx  1 root root    0 Jul 13 09:24 gpio44 -> ../../devices/platform/ocp/4804c000.gpio/gpio/gpio44
lrwxrwxrwx  1 root root    0 Jul 13 09:24 gpio45 -> ../../devices/platform/ocp/4804c000.gpio/gpio/gpio45
lrwxrwxrwx  1 root root    0 Jul 13 09:24 gpio46 -> ../../devices/platform/ocp/4804c000.gpio/gpio/gpio46
lrwxrwxrwx  1 root root    0 Jul 13 09:24 gpio47 -> ../../devices/platform/ocp/4804c000.gpio/gpio/gpio47
lrwxrwxrwx  1 root root    0 Jul 13 09:24 gpio48 -> ../../devices/platform/ocp/4804c000.gpio/gpio/gpio48
lrwxrwxrwx  1 root root    0 Jul 13 09:24 gpio49 -> ../../devices/platform/ocp/4804c000.gpio/gpio/gpio49
lrwxrwxrwx  1 root root    0 Jul 13 09:24 gpio5 -> ../../devices/platform/ocp/44e07000.gpio/gpio/gpio5
lrwxrwxrwx  1 root root    0 Jul 13 09:24 gpio50 -> ../../devices/platform/ocp/4804c000.gpio/gpio/gpio50
lrwxrwxrwx  1 root root    0 Jul 13 09:24 gpio51 -> ../../devices/platform/ocp/4804c000.gpio/gpio/gpio51
lrwxrwxrwx  1 root root    0 Jul 13 09:24 gpio60 -> ../../devices/platform/ocp/4804c000.gpio/gpio/gpio60
lrwxrwxrwx  1 root root    0 Jul 13 09:24 gpio61 -> ../../devices/platform/ocp/4804c000.gpio/gpio/gpio61
lrwxrwxrwx  1 root root    0 Jul 13 09:24 gpio65 -> ../../devices/platform/ocp/481ac000.gpio/gpio/gpio65
lrwxrwxrwx  1 root root    0 Jul 13 09:24 gpio66 -> ../../devices/platform/ocp/481ac000.gpio/gpio/gpio66
lrwxrwxrwx  1 root root    0 Jul 13 09:24 gpio67 -> ../../devices/platform/ocp/481ac000.gpio/gpio/gpio67
lrwxrwxrwx  1 root root    0 Jul 13 09:24 gpio68 -> ../../devices/platform/ocp/481ac000.gpio/gpio/gpio68
lrwxrwxrwx  1 root root    0 Jul 13 09:24 gpio69 -> ../../devices/platform/ocp/481ac000.gpio/gpio/gpio69
lrwxrwxrwx  1 root root    0 Jul 13 09:24 gpio7 -> ../../devices/platform/ocp/44e07000.gpio/gpio/gpio7
lrwxrwxrwx  1 root root    0 Jul 13 09:23 gpiochip0 -> ../../devices/platform/ocp/44e07000.gpio/gpio/gpiochip0
lrwxrwxrwx  1 root root    0 Jul 13 09:23 gpiochip32 -> ../../devices/platform/ocp/4804c000.gpio/gpio/gpiochip32
lrwxrwxrwx  1 root root    0 Jul 13 09:23 gpiochip64 -> ../../devices/platform/ocp/481ac000.gpio/gpio/gpiochip64
lrwxrwxrwx  1 root root    0 Jul 13 09:23 gpiochip96 -> ../../devices/platform/ocp/481ae000.gpio/gpio/gpiochip96
--w-------  1 root root 4096 Jul 13 09:23 unexport

But still I cannot run the epaper demos. Both gpio_test and epd_test fail at gpio setup since find_slots_and_ocp() fails to find "#define CAPE_MANAGER "bone_capemgr." and #define OCP "ocp." under /sys/devices:
```
$ ./driver-common/gpio_test 
gpio_test: GPIO_setup failed: Success
```

But then running any demo -> 
File "/home/debian/gratis/PlatformWithOS/demo/EPD.py", line 156, in _command
    f.write(c)
IOError: [Errno 103] Software caused connection abort

I can dig further but I did not really get what EPD assumes about the pin configuration.

hxw, 2015-07-14:
`$ config-pin overlay cape-universal` seems to be the key function, I think that the gpio.c needs to replicate the effect this command has.  I will have to study the config-pin script some more to see what this does.

hxw, 2015-07-14:
Just found that Arch dtc that supports overlays is optional package:
~~~
# pacman -S dtc-overlay
...this replaces original dtc
# dtc -@ -I dts -O dtb -o /lib/firmware/cape-universal-00A0.dtbo cu-00A0.dts
# echo cape-universal > /sys/devices/platform/bone_capemgr
-bash: echo: write error: Invalid argument
~~~
`dmesg` has this:
~~~
[457317.844257] bone_capemgr bone_capemgr: slot #42: 'Override Board Name,00A0,Override Manuf,cape-universal'
[457317.875862] of_resolve_phandles: Could not find symbol 'gpio4'
[457317.882153] bone_capemgr bone_capemgr: slot #42: Failed to resolve tree
~~~

I check the source the dts file seems to use gpio1..4 so rename 0..3
~~~
# cp cape-universal-00A0.dts cu-00A0.dts                                                                                                                              
# sed -i 's/gpio1/gpio0/g' cu-00A0.dts                                                                                                                                
# sed -i 's/gpio2/gpio1/g' cu-00A0.dts                                                                                                                                
# sed -i 's/gpio3/gpio2/g' cu-00A0.dts                                                                                                                                
# sed -i 's/gpio4/gpio3/g' cu-00A0.dts                                                                                                                                
# dtc -@ -I dts -O dtb -o /lib/firmware/cape-universal-00A0.dtbo cu-00A0.dts                                                                                          
# echo cape-universal > /sys/devices/platform/bone_capemgr/slots                                                                                                      
~~~
no error and spi appeared!
~~~
# ls -l /dev/*spi*
crw------- 1 root root 153, 1 Jul 14 18:04 /dev/spidev1.0
crw------- 1 root root 153, 0 Jul 14 18:04 /dev/spidev1.1
crw------- 1 root root 153, 3 Jul 14 18:04 /dev/spidev2.0
crw------- 1 root root 153, 2 Jul 14 18:04 /dev/spidev2.1
~~~

I still have to check if PWM is present - I did not see it a first glance, but maybe its name has changed

doganyazar, 2015-07-14:
Great! Let me know if you want me to try something :)

hxw, 2015-07-15:
Which version of the display do you have?  Only the V110G1 need PWM

doganyazar, 2015-07-15:
Yeah that is the one I am testing with at the moment. PANEL_VERSION='V110_G1'

hxw, 2015-07-16:
I have managed to get GPIO and PWM working on:
`Linux alarm 4.1.1-2-ARCH #1 Mon Jul 6 21:13:46 MDT 2015 armv7l GNU/Linux`

I used a slightly modified cape-universal device tree and a rewritten gpio.c these files are in the linux-4 subdirectory.  It is sufficient to get GPIO/PWM to work in gpio_test + 3 LEDs, but I think SPI is not yet enabled on my kernel.

hxw, 2015-07-16:
just tried this:
~~~
# echo spi > /sys/devices/platform/ocp/ocp:P9_17_pinmux/state
# echo spi > /sys/devices/platform/ocp/ocp:P9_21_pinmux/state
# echo spi > /sys/devices/platform/ocp/ocp:P9_22_pinmux/state              
# echo spi > /sys/devices/platform/ocp/ocp:P9_18_pinmux/state
~~~
now epd_test can work - need to figureout where code to do this should go

hxw, 2015-07-16:
tomorrow I'll look a Mkaefile see if I can detect linux version and select correct gpio.c

doganyazar, 2015-07-16:
Why can't you just use beaglebone-universal-io for all versions and let beaglebone-universal-io to decide which Linux kernel it is. And use your wrapper around config-pin for all versions ?
I see that in config-pin they already have logic to figure out which kernel it is:
```
#4.1.x where lots of things changed...
#use the new location of bone_capemgr for detection..
if [ -d /sys/devices/platform/bone_capemgr/ ] ; then
	OCPDIR=/sys/devices/platform/ocp/ocp*
	SLOTS=/sys/devices/platform/bone_capemgr/slots
fi
```

hxw, 2015-07-17:
It gave errors when trying to load the device tree, on my system and had to rename the gpios in it from1..4 to 0..3.  The kernel is 4.0 so maybe it was different to 4.1 the modified gpio.c uses the new locations and is somewhat simpler than the old one.  I was hoping that this device tree would converge across all ARM boards then there could just be one GPIO module that was universal. even across the BSDs (though they, FreeBSD at least, use /dev/gpio).

For this Arch system, I can make it work with the modified device tree, but it pre-exports all available GPIOs and there is no need to use /sys/class/gpio/export to enable them.  Without that device tree the default boot allows me to export/unexport GPIOs but no access to PWM,SPI or MUX.  Must have MUX access to switch pins from 'default' to 'pwm' or 'spi'.

If I remember correctly you are using Debian, perhaps they have slightly different kernel that makes config-pin work; I have not tried it on the most recent update, and I will try it again, I probably missed something the first time I tried.  (_It just occurred to me that previously I was trying with default dtc not the dtc-overlay version I dicovered later, so the universal-io might just work now with the latest gpio.c_)

My system is still running and the clock is display is updating nicely.  So we know it can work, just need to get it cleaned up  to use universal-io and portable across Arch and Debian.  Thanks for you patience while I muddle my way through this stuff and don't hesitate to send patches or other info.

doganyazar, 2015-07-17:
Thanks, good plan and great effort! :) Yeah I use debian. I am a bit unclear about the latest situation then. Is there a version I can try out now and try to help you out or should I just wait until you clean up and then give it a try?

hxw, 2015-07-18:
I have at the moment `BeagleBone/linux-4/gpio.c` which is the basis of the new driver.
in it is this definition:
~~~
// stand-alone file - diable all above firmware files
#define CAPE_UNIVERSAL "cape-universal"
~~~
If the universal-io works as I think then you could try copying `BeagleBone/linux-4/gpio.c` over `BeagleBone/gpio.c` modifying the string "cape-universal" in to match the universal-io DTBO file installed from universal-io.  Then recompiling.

If you hook up 3 LEDs and run the gpio test program, two LEDS should flash(GPIO) and one will vary in brightness(PWM).  At this point you would know if GPIO and PWM are operational.  If PWM is not running look for the string array: `const char *const pwm_state_file[] = {` and make sure the files are named correctly.

If the above works `ls -l /dev/spi*` to make sure the SPI is loaded also check that the SPI MUX files are correct.  They are near the gpio files, look for: `const char *const spi_state_file[] = {` if changes are required recompile, then try the epd test program.

(I will be travelling for the next month, so will probably not be able to try this un til mid August)

hxw, 2015-07-18:
Just to note the cape universal in the linux-4 is almost identical to the beaglebone-universal-io version; I rename gpio1..4 to gpio0..3 since these appear to be named 0..3 on Arch kernel.

doganyazar, 2015-07-20:
Thank you very much! I will try these steps now and see how it goes. Have fun traveling :)

doganyazar, 2015-07-26:
I compiled it with linux-4/gpio.c and after enabling universal overlay EPD test program worked fine. I can see different images are loaded on epaper. 
And tried DrawDemo.py as well and it also worked fine after make install. 

So all good on my side! Thanks!

hxw, 2015-09-05:
closing as appears to be fixed

closed 2015-09-05 by hxw