repaper/gratis #24: No activity on the EPD screen

yskkin, 2015-05-22:
I'm using Raspberri Pi 2 Model B and 2.7" EPD.

compilation goes fine, epd-fuse service is running and /dev/epd is mounted but demo/DrawDemo.py and other demo codes does not show anything on the screen.

I checked [this FAQ page](https://learn.adafruit.com/repaper-eink-development-board-arm-linux-raspberry-pi-beagle-bone-black/faq) and tried both of `make PANEL_VERSION=V230_G2 rpi` and `make PANEL_VERSION=V231_G2 rpi`. Neither of them work. 

output of `uname -a` is
```
Linux raspberrypi 3.18.13-v7+ #785 SMP PREEMPT Mon May 18 17:53:02 BST 2015 armv7l GNU/Linux
```

yskkin, 2015-05-29:
I can run epd_test with the following modification. 

```
diff --git a/PlatformWithOS/RaspberryPi/gpio.c b/PlatformWithOS/RaspberryPi/gpio.c
index 2048a4f..36128b4 100644
--- a/PlatformWithOS/RaspberryPi/gpio.c
+++ b/PlatformWithOS/RaspberryPi/gpio.c
@@ -34,7 +34,7 @@

 // register addresses in Rasberry PI
 enum {
-       BCM_PERIPERALS_ADDRESS = 0x20000000,
+       BCM_PERIPERALS_ADDRESS = 0x3F000000,
        GPIO_REGISTERS         = 0x00200000,
        PWM_REGISTERS          = 0x0020C000,
        CLOCK_REGISTERS        = 0x00101000,
```

However, epd-fuse is not working fine yet.

hxw, 2015-06-01:
I did not try any testing with B2, and only recently have one.  I Thought that changing the base address should be sufficient.   So the program works correctly with this change?

Any Idea if this can be determined at run time?  (read /proc/cpuinfo perhaps)

hxw, 2015-06-01:
fix applied: 08e936eee1d

closed 2015-06-01 by hxw