repaper/gratis #26: Provide Implementation for CC3200 and MSP432 LaunchPads

rei-vilo, 2015-07-17:
Support for the CC3200 is provided only with CCS6, and not with Energia.

Energia now bring RTOS for the CC3200 and MSP432 LaunchPads.

Please adapt the libraries for the CC3200 and MSP432 LaunchPads on Energia Multi-Tasking.

rei-vilo, 2015-08-02:
Results on LaunchPad 
* Works on MSP430F5529, LM4F120, TM4C123, TM4C129
* Fails on MSP432 and CC3200

See commit d9cbf4a and pull-request #30 LaunchPad MSP430F5529, CC3200 and MSP432

rei-vilo, 2015-08-04:
Is the issue with MSP432 linked to [MSP432 GPIO with High-Drive capability](http://e2e.ti.com/support/microcontrollers/msp430/f/166/t/428632) limited to `P2.0` `P2.1` `P2.2` `P2.3`?

hxw, 2015-09-20:
The V231_G2 now supports MSP432P401R via the MSP-EXP board.

There is one problem though, the repaper board cannot be directly mounted as the PANEL_ON pin which would be connected to P3_6 cannot be driven high.  In fact after a `SPI.begin()` call it seems that UCB2 is activated and pins P3_5, P3_5 and P3_7 are all assigned to this peripheral.

As a workaround the refactored I/O library assing the PANEL_On to P5_1 on this board, so use a cable connection that connects the PANEL_ON(Red,11) to P5_1 on second row of pins; rthe remaining wires are to their original pins.

hxw, 2015-09-21:
I forgot to mention the SPI has to be in mode3 on the MSP432, unlike the Arduino and MSP430 which bot use mode0.  Is the mode->pol/pha mapping correct in MSP432 driver?

rei-vilo, 2016-01-10:
I managed to get it work on CC3200 EMT, after fixing 

* the [#822](https://github.com/energia/Energia/issues/822) **CC3200: Inconsistent Resolutions Between Non-EMT and EMT** with this modification on `S5813A.cpp`.

```
long S5813A_Class::readVoltage(void) {
	long vADC = analogRead(this->temperature_pin);

// Added for CC3200 EMT    
#if defined(ENERGIA_MT)
#if defined(__CC3200R1M1RGC__) || defined(__CC3200R1MXRGCR__)
    vADC *= 4;
#endif
#endif
// End
/*
    Serial.print("analogRead=");
    Serial.print(vADC);
    Serial.print("\treadVoltage=");
    Serial.println(REV_PD((vADC * ADC_MAXIMUM_mV) / ADC_COUNTS));
 */
	return REV_PD((vADC * ADC_MAXIMUM_mV) / ADC_COUNTS);
}
```

* and adding a dummy `SPI.begin()` on `EPD_V231_G2.cpp`.

```
static void SPI_on(void) {
    
    // Energia EMT freezes on SPI.end() if no previous SPI.begin()
    // Dummy SPI.begin()
    SPI.begin();
    
	SPI.end();
	SPI.begin();
	SPI.setBitOrder(MSBFIRST);
...
```

Still no luck on the MSP432.

>
Temperature = 19 Celsius
dc_state = 0x0
EPD: DC failed
>

I'm going to test `mode 3`.

rei-vilo, 2016-01-10:
SPI `mode 3` is indeed activated in MSP432. Still no display.