repaper/gratis #27: #include <{% DRIVER:header %}>

qhn0511, 2015-07-18:
Need help in figuring out what the error message below means which is generated from Arduino Sketch after the following two lines of code.

           #include <{% DRIVER:header %}>
           #define SCREEN_SIZE {% PANEL:size %}

"demo.ino:47:31: fatal error: {% DRIVER:header %}: No such file or directory
compilation terminated."

hxw, 2015-07-23:
The sample files have been changed to templates, and after any changes I run s script that generates examples for each supported display size and panel driver chip.

Don't try to load these fiels directly instead goto to the library and choose from the example menu and you will see all the correct files listed: The menu sequence for this is:

`File → Examples →EPD_Vxxx_Gy →demo_sss`
```
xxx = panel version: 110, 230 or 231
y   = Chip number: 1 or 2
sss = panelsize: 144, 200 or 270
```
( if the EPD_... entries are not present make sure settings sjketchbook location is correct then restart Arduino IDE )

Hope that helps

qhn0511, 2015-07-23:
Thank You!

It all make sense now.

Quan Nguyen