repaper/gratis #31: error: overflow in constant expression and size of array 'new_image' is negative

rei-vilo, 2015-08-21:
Change lines 36 and 38 for 

```
#if EPD_IMAGE_TWO_ARG
	uint8_t old_image[(uint32_t)pixel_width * pixel_height / 8];
#endif
	uint8_t new_image[(uint32_t)pixel_width * pixel_height / 8];
```

Report log

>
In file included from thermo_270.ino:45:0:
~/Documents/Projets/Arduino/libraries/EPD_GFX/EPD_GFX.h:36:50: error: overflow in constant expression
  uint8_t old_image[pixel_width * pixel_height / 8];
                                                  ^
~/Documents/Projets/Arduino/libraries/EPD_GFX/EPD_GFX.h:36:50: error: size of array 'old_image' is negative
~/Documents/Projets/Arduino/libraries/EPD_GFX/EPD_GFX.h:38:50: error: overflow in constant expression
  uint8_t new_image[pixel_width * pixel_height / 8];
                                                  ^
~/Documents/Projets/Arduino/libraries/EPD_GFX/EPD_GFX.h:38:50: error: size of array 'new_image' is negative
Error compiling.
>

rei-vilo, 2015-08-21:
Similarly, with a 12-bit DAC, `vADC * ADC_MAXIMUM_uV` throws an overflow.

Change line 114 of **S5813A.cpp** for 

```
long S5813A_Class::readVoltage(void) 
{   
    long vADC = analogRead(this->temperature_pin);
    return REV_PD(((uint32_t)vADC * ADC_MAXIMUM_uV) / ADC_COUNTS);
}
```

ecdr, 2015-08-21:
One solution is to add a scale factor, defined based on platform.

// Divide by scale to prevent overflow, then multiply to restore range
	return REV_PD(((vADC * (ADC_MAXIMUM_uV/SCALE)) / ADC_COUNTS)*SCALE);

Define SCALE appropriately depending on the ADC width (1 for arduino)

For Tiva:

// Tiva LaunchPads / TI TM4C123x, TM4C129x run at 3.3V
#define PIN_TEMPERATURE  A8
// pin A8 on Connected LP upper, A6 on Connected LP lower

// Energia does not define an analog reference name for Tiva
#define ANALOG_REFERENCE 0

// ADC maximum voltage at counts
#define ADC_MAXIMUM_uV   3300000L
#define ADC_COUNTS       4096L

// Scale factor, to prevent overflow
#define SCALE 100

I have done a port to get gratis running on the Tiva/Stellaris launchpads, which includes this fix, but looks like I didn't push it to my account yet.  Will try to get it up soon.

hxw, 2016-02-21:
I think this is fixed

closed 2016-02-21 by hxw