repaper/gratis #32: amslide won't work on 2.7

kevinmaeeweb, 2015-08-31:
hi, i'm actually doing a project that is the same as how the amslide would work. im using an epd extension board to display multiple images. i was hoping to load them from an sd card shield. im using arduino uno for my interface between the sd and the display board. i have tried many configurations already, but to no avail im having issues with SPI. i have tried using the recently posted code in this site. independently, the uploading the demo code 270 in version 230 worked fine. also for the sd card. but when they are both connected. i cant seem to solve the problem.. something on pin 12? if i remove it, the display would work but SD won't. the setup is actually mounted on top of each other.

any help would be appreciated. 

thanks,
kevin

hxw, 2015-09-01:
you will need to check that the CS for SD and CD for EPD are on different pins.  If you are not using the SPI FLASH on the eval board the connect this th 3.3V to siaple this flash chip and free up one I/O.

kevinmaeeweb, 2015-09-01:
Hi, me and my colleagues have already checked and assigned different pins for the CS of the SD and EPD. Just to clarify, do you mean that the SPI Flash pin will be connected to 3.3V. the amslide code was given that the flash is instantiated on pin 8. what would happen to the free I/o? would that be the one used for the SD CS? i do appreciate your response.

hxw, 2015-09-02:
I was thinking that the FLASH CS could be used as SD CS if you were short of I/O pins. Tying FLASH CS high just disables it (assuming it is not being used) to free up the I/O (also remove any flash setup from the amslide.ino)

Did you check that set_spi_for_sdcard is correct for you board?

hxw, 2015-09-02:
Which 2.7 panel version/COG version do you have?

kevinmaeeweb, 2015-09-02:
Hi, what i have here is a COG 2.7 panel. im using an arduino uno and a SD shield from egizmo.

"Did you check that set_spi_for_sdcard is correct for you board?"--about this, i haven't check this though. how would i know if this the right for my application?

sorry, i really don't have much knowledge about this stuff. thanks for your patience..

kevinmaeeweb, 2015-09-02:
[image]
[image]
[image]

i took pictures of my setup if it would help. 

thanks

hxw, 2015-09-02:
The SD card CS is part of the FAT FS configuration as in:
```
        pinMode(Pin_SD_CS, OUTPUT);
        if (!SD.begin(Pin_SD_CS)) {
```

hxw, 2015-09-02:
`boolean begin(uint8_t csPin = SD_CHIP_SELECT_PIN);` is the default from your boards library.  It looks from you photo that the CS1 jumper is set which puts the SD CS1 on I/O 4.  maybe better to remove solder from this and wire CS1 to the pin.  I think the CD1 pin need to be left open since it has pull down.

hxw, 2015-09-05:
I have just checked the amslide with the original aLaMode board to make sure that it does still work correctly.  I did add some #if so I could use the V231_G2 2.7" panel that I have.  I is currently working, reading images from a 4GB uSD Card.

kevinmaeeweb, 2015-09-08:
Hi, 
Sorry I have been busy lately. If it is ok to you, Can you show me your coding? The ones you said you added the "if's". I still don't have any progress here. Also, I'm curious, will the amslide still work even i don't use the aLaMode board? I checked and email, wyolum about this. But, it there wasn't much information. Thanks

kevinmaeeweb, 2015-09-08:
I also tried uploading the amslide you recently updated. It wouldn't work also. I believe the connection here is the problem now. Its either the arduino or the sd shield is not working well.. or maybe my connection.

kevinmaeeweb, 2015-09-11:
May I ask something, the documentation of  the amslide says that it was done using the alamode board but it also states there that it should work on arduinos that have an sd or plugin shield. for my sd shield the default is pin 4. in the coding, i noticed that the pin discharge is also using this pin. does that mean i can use the pin 4 for the CS for the sd? also, in the code, it states there that the epd flash cs is pin 9 and epd cs is pin 8. if i understood it right, im all good if i use pin 10 for the sd cs. im really having a hard time here. i have tried removing the solder of the cs in the sd shield and conected it directly to the pin 4. any help would be deeply appreciated.

hxw, 2015-09-14:
if you have to use pin 4 as you CS then change the definition in the amdemo.ino  these are the defaults I set:
~~~
// Arduino IO layout
const int Pin_TEMPERATURE = A0;
const int Pin_PANEL_ON = 2;
const int Pin_BORDER = 3;
const int Pin_DISCHARGE = 4;
const int Pin_PWM = 5;
const int Pin_RESET = 6;
const int Pin_BUSY = 7;
const int Pin_EPD_CS = 8;
const int Pin_EPD_FLASH_CS = 9;
const int Pin_SD_CS = 10;
~~~

you could swap with 10 and make 10 the discharge, thus freeing pin 4 for you SD CS

kevinmaeeweb, 2015-09-14:
hi, i have tried using a different sd shield, an audio shield exactly.it worked fine now. but id like to try this configurations you sent. maybe the sd shield wouldn't be wasted after all. thanks

kevinmaeeweb, 2015-09-14:
hi, i know this is out of context. but do you know how to make custom images with the ".27" format?. now im trying to make multiple images to run in the amslide. also, could you explain what the text in the index.txt mean? it says there that 
5 aphro.27
6 venus.27
what those this 5 and 6 mean. and all the  other numbering. any help would help.thanks

kevinmaeeweb, 2015-09-15:
hi, on the creating the image. i was able to make 10 new images that i customize to be displayed on the board. the problem now is that i dont know how it will be numbered in the index.txt file. it there a necessary sequence for this? or does the numbers really matter? thanks

hxw, 2015-09-16:
The file format is:
~~~
<integer-seconds>  <space> <fat-16-file.ext> <new-line>
~~~
The file is read and displayed, there is a for loop to execute the delay, the next index line is read and at the end of the index it is reopened and the images are displayed again.

The format of the image file exactly matches the display byte count.  I did not try to do any scaling or have any header, it is just raw binary bytes that are put straigh onto the display.  If you look at the sample XBM image files (e.g.`/Sketches/libraries/Images/cat_2_0.xbm` you will observe that they are just a C header with hex data.   The Linux shell script `convert_xbm_to_binary.sh` takes the XBM strips the C text and convert the file to binary data.  A c program that included the XBM and just used `fwrite(cat_2_0_bits, sizeof(cat_2_0_bits), 1, fp)`  would achieve the same effect - hope that is clear.

I made the images originally using GIMP; the process was rotate/resize/crop image to best fit the display size, convert to 2 colour black/white export as XBM.

kevinmaeeweb, 2015-09-16:
Hi, thank you very much for the help. there is some things i recognized in the for loop,it is an infinite loop and that the condition in the if is that, it opens the index.txt file den closes it. is there a way we could just reset to the starting line if there is no more files to display. i believe the syntax ".read()" is a line by line reading of the file. also, this <integer-seconds> is the time of the delay for the next image? im very sorry if i have a lot of questions. i really just want to make this work. thanks and great work there.

hxw, 2015-09-17:
have a look to see if there is a "seek" function for the file, if there is then seek to position zero and the next read would fetch the first line again.

You could always try another method:  like name all the files as numbers `0000.img`, `0001.img` etc. then simply have a counter and use: `snprintf(buffer, sizeof(buffer), "%04d.img", counter)` to make the file names.  You would probably have a constant delay in this case.

kevinmaeeweb, 2015-09-17:
A quick look on the code. it has seek function. but im not quite sure what you mean by seek function for the file. but for now, im still stuck with the images not looping back and forth. and only the second to the last image will be display. anyhow, im quite preoccupied today, but i'll try this later. Thanks.

hxw, 2015-09-17:
trye `index_file.seek(0)` instead of closing and reaopening

kevinmaeeweb, 2015-09-17:
I tried to have some test points in the code. it code to an infinite loop when it enters:
// skip to next line
	while (0x0a != c) {
        Serial.println("inside 0x0a");
		c = index_file.read();
	}
i removed the while loop an maintained the c = index_file.read(); now its looping and working fine. Ill be having testings at this time. Thanks and good work.