repaper/gratis #38: spi-bcm2835 on Raspberry Pi 2

delormev, 2016-01-17:
It looks like my device is using spi-bcm2835 rather than spi-bcm2708.

In https://github.com/repaper/gratis/blob/master/PlatformWithOS/RaspberryPi/epd-fuse.sh#L34, changing `modprobe spi-bcm2708` to `modprobe spi-bcm2835` (line ) did the trick for me, for I'm guess it depends on the model / version of the OS people are using...

shawaj, 2016-01-17:
I think spi-bcm2835 is correct on all versions of Raspberry Pi now. I will post on RasPi forum to confirm and then post back here

shawaj, 2016-01-18:
So far it looks like spi-bcm2708 is completely deprecated and Raspbian now uses spi-bcm2835 - https://www.raspberrypi.org/forums/viewtopic.php?f=44&t=133024

hxw, 2016-02-21:
should be fixed

closed 2016-02-21 by hxw