repaper/gratis #39: Partial refresh error!

go2srinidhi, 2016-01-27:
Using 1.44" V231_G2 display. 
Am interested to know how i would achieve partial updates as demonstrated on your Youtube video. 
Trying to use the python examples like "DrawDemo.py" or "PartialDemo.py" or "CounterDemo.py" but i get errors as shown below:

root@beaglebone:/home/debian/gratis/PlatformWithOS# python demo/PartialDemo.py 3 20
panel = EPD 1.44 128 x 96  version=4 COG=2 FILM=231
Traceback (most recent call last):
  File "demo/PartialDemo.py", line 89, in <module>
    main(sys.argv[1:])
  File "demo/PartialDemo.py", line 52, in main
    demo(epd, objects, frames)
  File "demo/PartialDemo.py", line 82, in demo
    epd.display(image)
  File "/home/debian/gratis/PlatformWithOS/demo/EPD.py", line 139, in display
    f.write(image.tobytes())
  File "/usr/lib/python2.7/dist-packages/PIL/Image.py", line 512, in __getattr__
    raise AttributeError(name)
AttributeError: tobytes

I would like to achieve similar output as shown in the video. 
https://www.youtube.com/watch?v=enzUbiSWenQ

hxw, 2016-01-28:
It looks like you are using an older version of Python currently the code runs on 2.7.11.
If you can please update ad you should be ok, if you cannot update and need to continue with older Python version then you will need to change one line in the `demo/EPD.py` file

Modify line 194 which reads: `f.write(image.tobytes())` to be: `f.write(image.tostring())`

go2srinidhi, 2016-01-28:
Thanks,
I did modify as u suggested. No changes on display. :( But the code isn't throwing any errors.

hxw, 2016-02-01:
I am having problems with BBB, I updated and it no longer boots, I will recover an older image and try that

hxw, 2016-02-01:
I put the old image at:
https://drive.google.com/file/d/0ByAJ8vCbYBDgTTNLcUVjZVlJcnc/view?usp=sharing

dowload and name it as `bbb-1.44.img.xz` then extract to an SD card:
```
$ xz -d < bbb-1.44.img.xz | dd bs=64M of=/dev/sdX
```
replace `/dev/sdX` with the right drive to access the USB card writer (be careful!)

The image is 2GB so the card need to be at least this size.

Note: SHA256 of image and its compressed version:
````
fae9e587a9b51092d05613479aab7a5337b5ab097975851c0f5b13151336362d  bbb-1.44.img
4593775fb20ddd7466c1e33edc709600b7f63c93a726b2036be0b2c20c1bad0c  bbb-1.44.img.xz
````

hxw, 2016-02-01:
I have a problem on first start, the epd-fuse fails when counter demo is started  but restarting it once seems to be enough, maybe I hav it starting too ealy in the boot sequence.
```
$ python2 CounterDemo.py 1234
# failure here
$ sudo systemctl restart epd-fuse
$ python2 CounterDemo.py 1234
```

hxw, 2016-02-01:
Please try if possible it will determine if your hardware is working.

I have tried to update this image, but the latest kernel hangs, even though I removed the device tree file before updating.

go2srinidhi, 2016-02-01:
Thanks for the suggestion.
I shall try these things and post the update asap. And also in the near future, i would like to port it to kernel 4.1.13 as my target hardware runs this version.

hxw, 2016-02-02:
Maybe will need to update the device tree file for newer kernel, I used cape universal in this old image and the file is: `/lib/firmware/cape-universal-00A0.dtbo`

go2srinidhi, 2016-02-04:
Thanks,
Am taking care of the device tree stuffs. Anything else in specific to look into?

hxw, 2016-02-05:
The only thing seems to be that epd-fuse is crashing on first use, not sure why possible systemd service starts it at the wrong time (too early?).  Also need to set the service file to restart it on failure.  I am using Arch so not sure if this applies to your system (perhaps you use sysvinit or upstart)