repaper/gratis #42: Error on Raspberry Pi, FAQ appears out of date

peterhinch, 2016-03-21:
I have (recent, 2.7 inch) COG 2 panels. I've tried two known good EPD's, one from Adafruit, one from EmbeddedArtists and two Pi's (a model B and a B+) running a recent Raspbian. I did (as root)
```
apt-get update
apt-get upgrade
```
before starting on the installation instructions. I get the following when I run the first demo:
```
pi@raspberrypi:~/gratis/PlatformWithOS $ python demo/DrawDemo.py
panel = EPD 2.7 264 x 176  version=4 COG=2 FILM=231
Traceback (most recent call last):
  File "demo/DrawDemo.py", line 70, in <module>
    main(sys.argv[1:])
  File "demo/DrawDemo.py", line 31, in main
    epd.clear()
  File "/home/pi/gratis/PlatformWithOS/demo/EPD.py", line 152, in clear
    self._command('C')
  File "/home/pi/gratis/PlatformWithOS/demo/EPD.py", line 156, in _command
    f.write(c)
IOError: [Errno 103] Software caused connection abort
```
The build instructions in the FAQ don't work. I built (as root) with
```
make rpi PANEL_VERSION='V231_G2'
make rpi-install PANEL_VERSION='V231_G2'
```
The build looked OK but install produced the following output:
```
root@raspberrypi:/home/pi/gratis/PlatformWithOS# make rpi-install PANEL_VERSION='V231_G2'
make PLATFORM=../RaspberryPi PANEL_VERSION="V231_G2" EPD_IO="epd_io.h" -C driver-common install
make[1]: Entering directory '/home/pi/gratis/PlatformWithOS/driver-common'
[ -e epd_fuse ] || false
install --group=root --mode=750 --owner=root epd_fuse "/usr/sbin"
[ -e /etc/default/epd-fuse ] || install --group=root --mode=644 --owner=root epd-fuse.default /etc/default/epd-fuse
if [ -e "../RaspberryPi/epd-fuse.conf" ] ; \
then \
  install --group=root --mode=644 --owner=root "../RaspberryPi/epd-fuse.conf" /etc/init ; \
fi
if [ -e "../RaspberryPi/epd-fuse.sh" ] ; \
then \
  install --group=root --mode=755 --owner=root "../RaspberryPi/epd-fuse.sh" /etc/init.d/epd-fuse ; \
  update-rc.d epd-fuse defaults ; \
fi
make[1]: Leaving directory '/home/pi/gratis/PlatformWithOS/driver-common'
```
Is this correct? Any assistance gratefully received.

hxw, 2016-03-22:
if you restart the epd fuse (`service epd-fuse restart`) are there any files in /dev/epd?

if so what is the output of:
```
cat /dev/epd/panel
cat /dev/epd/version
```

peterhinch, 2016-03-22:
```
pi@raspberrypi:~ $ sudo -i
root@raspberrypi:~# service epd-fuse restart
root@raspberrypi:~# cat /dev/epd/panel
EPD 2.7 264x176 COG 2 FILM 231
root@raspberrypi:~# cat /dev/epd/version
4
root@raspberrypi:~# 
```
Thank you for your help with this!

hxw, 2016-03-22:
see if clearing the display works: `echo C > /dev/epd/command`
or does this cause the epd-fuse to crash? if so, dor `dmesg` show eny errors?
(I have had epd-fuse crash on first access immediately after a reboot and restarting it allowed demod to run, but that was on Arch)

peterhinch, 2016-03-22:
```
root@raspberrypi:~# echo C > /dev/epd/command
-bash: echo: write error: Software caused connection abort
root@raspberrypi:~# 
```
There is nothing new in dmesg, these are the last entries (after boot 1hr 40 mins ago):
```
[   16.513688] Adding 102396k swap on /var/swap.  Priority:-1 extents:1 across:102396k SSFS
[   18.731149] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[   18.732742] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[   20.323702] smsc95xx 1-1.1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0xCDE1
[   20.350321] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   49.556833] EXT4-fs (mmcblk0p5): mounted filesystem with ordered data mode. Opts: (null)
root@raspberrypi:~# 
```

hxw, 2016-03-22:
could you run the epd-fuse program in another terminal and try the echo to see if any output to stderr
```
service epd-fuse stop
/usr/sbin/epd_fuse --panel=2.7 -o allow_other -o default_permissions /dev/epd
```

Also what is the output of `uname -a` as it looks like I will have to try and build a raspian image.
So I can match the same kernel.

peterhinch, 2016-03-22:
Here is the session:
```
pi@raspberrypi:~ $ sudo -i
root@raspberrypi:~# service epd-fuse stop
root@raspberrypi:~# /usr/sbin/epd_fuse --panel=2.7 -o allow_other -o default_permissions /dev/epd
root@raspberrypi:~# uname -a
Linux raspberrypi 4.1.19+ #853 Wed Mar 9 18:04:48 GMT 2016 armv6l GNU/Linux
root@raspberrypi:~# 
```
I rebooted after the last error, so this session was on a freshly booted Pi.

hxw, 2016-03-24:
can you confirm is spi device loaded?
`lsmod` should list the driver: `spi_bcm2835`
and:
```
$ ls  -l /dev/spi*
crw-rw---- 1 root spi 153, 0 Mar 24 05:56 /dev/spidev0.0
crw-rw---- 1 root spi 153, 1 Mar 24 05:56 /dev/spidev0.1
```

if not check that `/boot/config.text` has the following line enabled:
(the orginal line had `#` in front)
```
dtparam=spi=on
```
if it was commented remove the `#` save  file and reboot.

peterhinch, 2016-03-24:
``/boot/config.text`` did have that line commented out. Demo now works - thanks very much for the assistance.

I've not closed the issue in case you want to update the FAQ.

hxw, 2016-03-25:
The rasbian note is in: 9c38488

closed 2016-03-25 by hxw