repaper/gratis #45: Support for 10.2" Pervasive Displays (EZ102CT011)

edundias, 2016-05-16:
Hi,

I'm trying to connect a 10.2" display from Pervasive Displays (model EZ102CT011) and when I use the epd-fuse and set EPD_SIZE to 10.2 it does not work. 
I haven't been able to display anything. Can I request some support please?
Thank you very much in advance.

shawaj, 2016-06-13:
As far as I know, this code will only support screen sizes from 1.44" up to 2.71 inch listed here - http://www.pervasivedisplays.com/products

hxw, 2016-06-14:
The open source code is limited to panels smaller than three inches owing to Patent and/or NDA requirements for datashets and timing patterns.
Therfore, please contact Pervasive directly to ask about any code or eval kits for greater than 2.7" displays.

shawaj, 2016-06-14:
@hxw do you know if there are any future plans to support open source on the larger ones?

repaper, 2016-06-15:
Like @hxw mentioned, the Repaper.org will focus on sub 3 inch open source only and always. Please contact Pervasive Displays for 4 inch above.