repaper/gratis #48: Implement partial update for MCUs

andrey-gvrd, 2016-07-21:
I noticed that partial update functionality is implemented only for PlatformsWIthOS. Is there a reason it's not done for MSP430 and Arduino?

Judging by the comments [here](https://www.youtube.com/watch?v=enzUbiSWenQ) It seems that the port already exists, but isn't released to this repository.

repaper, 2016-07-22:
The sample code is managed by Pervasive Displays officially on TI CCS project.  You can leave your message to the [Technical Support](http://www.pervasivedisplays.com/contact/technical-support) webform to get the source.
The [AdaFruit GFX](https://learn.adafruit.com/repaper-eink-development-board/libraries-and-examples) has partial update example as well.

andrey-gvrd, 2016-07-22:
I can't see partial update code in Adafruit's libraries. Are you referring to `EPD.image(IMAGE_1_BITS, IMAGE_2_BITS)`?

repaper, 2016-07-22:
https://github.com/repaper/gratis/issues/46

andrey-gvrd, 2016-07-22:
Are you referring to `EPD_GFX`? I see that it does use `this->EPD.image_sram(this->old_image, this->new_image)` in its `display()` method, but updating between screens still takes 3 seconds. What am I missing here?