repaper/gratis #51: how do we build on raspberry pi?

shawaj, 2016-09-05:
How do we build on raspberry pi after this recent change?

https://github.com/repaper/gratis/commit/5c5debe5de40178874d00f419bf02e008873b042

Gives error:

Cloning into 'gratis'...
remote: Counting objects: 2492, done.
remote: Total 2492 (delta 0), reused 0 (delta 0), pack-reused 2491
Receiving objects: 100% (2492/2492), 8.30 MiB | 1.09 MiB/s, done.
Resolving deltas: 100% (1291/1291), done.
Checking connectivity... done.
make: *** No rule to make target 'rpi'.  Stop.
make: *** No rule to make target 'rpi-install'.  Stop.
Failed to start epd-fuse.service: Unit epd-fuse.service failed to load: No such file or directory.

magnets99, 2016-09-06:
i'm glad it's not just me.
the adafruit tutorial also makes no sense now.
https://forums.adafruit.com/viewtopic.php?f=47&t=102608&p=514370#p514370

adafruit are referring people here, but this documentation is also wrong.

hxw, 2016-09-12:
I changed the Makefile a little to set better defaults and fix some mkdir bugs in it

shawaj, 2016-09-14:
So it works with RasPi now?

hxw, 2016-09-20:
I was able to compile and run withe the following:
```
cd gratis
make rpi EPD_IO=epd_io_free_uart.h PANEL_VERSION='V231_G2'
sudo make rpi-install EPD_IO=epd_io_free_uart.h PANEL_VERSION='V231_G2'
```
I used the alternate header because I built a new cable with a single large connector and the console port attached to USB serial rather than the separate wires cable.

the Rpi V2B is running arch Linux:
```
Linux 4.4.20-1-ARCH #1 SMP Wed Sep 7 19:10:13 MDT 2016 armv7l GNU/Linux
```

hxw, 2016-09-21:
I think this is ok now, if not please open new issue.

closed 2016-09-21 by hxw

francesco-vannini, 2017-03-15:
Me and @shawaj have been testing this. It seems that there is an error in the makefile? It compiles correctly now, but we get the following error when trying to start the service:

```Job for epd-fuse.service failed. See 'systemctl status epd-fuse.service' and 'journalctl -xn' for details.
```

When checking `systemctl status epd-fuse.service` we get the following errors:

```● epd-fuse.service - A Fuse driver for the Repaper.org EPD
   Loaded: loaded (/usr/lib/systemd/system/epd-fuse.service; enabled)
   Active: failed (Result: exit-code) since Wed 2017-03-15 18:11:20 UTC; 10s ago
  Process: 3592 ExecStartPre=/usr/bin/mkdir -p ${EPD_MOUNTPOINT} (code=exited, status=203/EXEC)

Mar 15 18:11:20 raspberrypi systemd[1]: epd-fuse.service: control process exited, code=exited status=203
Mar 15 18:11:20 raspberrypi systemd[1]: Failed to start A Fuse driver for the Repaper.org EPD.
Mar 15 18:11:20 raspberrypi systemd[1]: Unit epd-fuse.service entered failed state.
```
Any further thoughts?