repaper/gratis #53: Error in epd-fuse.service

NicolaP8, 2016-09-18:
Hi,
the file epd-fuse.service contains at line 13:
`ExecStartPre=/usr/bin/mkdir -p ${EPD_MOUNTPOINT} `
but this is wrong because you can never be sure of the position of mkdir (in my system is /bin). For this exists the PATH.

hxw, 2016-09-19:
Problem is that systemd seem to require absolute paths for the executable files :(
```
[/usr/lib/systemd/system/epd-fuse.service:13] Executable path is not absolute, ignoring: mkdir -p ${EPD_MOUNTPOINT}
```

hxw, 2016-09-20:
Maybe the best solution is to try to detect the programs during make install and used sed to modify the service file.  Any better solutions?

francesco-vannini, 2017-03-15:
We are also are seeing the same issue.

PR #65 (2017-03-22): Updated the mkdir and umount path
Updated the mkdir and umount path to /usr/bin/env <command>

This fixes #53 #61
closed 2017-03-22 by hxw