repaper/gratis #55: not enough space on Mega2560

shirish47, 2017-01-31:
I am using pervasive display 2.7 inch thermo_270.ino for EPD_231_G2 I get following error is: 

`"/private/var/folders/w2/jb9h2v3j5s5bs70q009xq2x00000gn/T/AppTranslocation/AA37A472-D2D8-4297-9EF6-726E8537F2C2/d/Arduino-2.app/Contents/Java/hardware/tools/avr/bin/avr-gcc" -w -Os -g -flto -fuse-linker-plugin -Wl,--gc-sections,--relax -mmcu=atmega2560  -o "/var/folders/w2/jb9h2v3j5s5bs70q009xq2x00000gn/T/arduino_build_643615/thermo_270.ino.elf" "/var/folders/w2/jb9h2v3j5s5bs70q009xq2x00000gn/T/arduino_build_643615/sketch/thermo_270.ino.cpp.o" "/var/folders/w2/jb9h2v3j5s5bs70q009xq2x00000gn/T/arduino_build_643615/libraries/SPI/SPI.cpp.o" "/var/folders/w2/jb9h2v3j5s5bs70q009xq2x00000gn/T/arduino_build_643615/libraries/EPD_FLASH/EPD_FLASH.cpp.o" "/var/folders/w2/jb9h2v3j5s5bs70q009xq2x00000gn/T/arduino_build_643615/libraries/EPD_V231_G2/EPD_V231_G2.cpp.o" "/var/folders/w2/jb9h2v3j5s5bs70q009xq2x00000gn/T/arduino_build_643615/libraries/S5813A/S5813A.cpp.o" "/var/folders/w2/jb9h2v3j5s5bs70q009xq2x00000gn/T/arduino_build_643615/libraries/Adafruit_GFX_Library/glcdfont.c.o" "/var/folders/w2/jb9h2v3j5s5bs70q009xq2x00000gn/T/arduino_build_643615/libraries/Adafruit_GFX_Library/Adafruit_GFX.cpp.o" "/var/folders/w2/jb9h2v3j5s5bs70q009xq2x00000gn/T/arduino_build_643615/core/core.a" "-L/var/folders/w2/jb9h2v3j5s5bs70q009xq2x00000gn/T/arduino_build_643615" -lm
"/private/var/folders/w2/jb9h2v3j5s5bs70q009xq2x00000gn/T/AppTranslocation/AA37A472-D2D8-4297-9EF6-726E8537F2C2/d/Arduino-2.app/Contents/Java/hardware/tools/avr/bin/avr-objcopy" -O ihex -j .eeprom --set-section-flags=.eeprom=alloc,load --no-change-warnings --change-section-lma .eeprom=0  "/var/folders/w2/jb9h2v3j5s5bs70q009xq2x00000gn/T/arduino_build_643615/thermo_270.ino.elf" "/var/folders/w2/jb9h2v3j5s5bs70q009xq2x00000gn/T/arduino_build_643615/thermo_270.ino.eep"
"/private/var/folders/w2/jb9h2v3j5s5bs70q009xq2x00000gn/T/AppTranslocation/AA37A472-D2D8-4297-9EF6-726E8537F2C2/d/Arduino-2.app/Contents/Java/hardware/tools/avr/bin/avr-objcopy" -O ihex -R .eeprom  "/var/folders/w2/jb9h2v3j5s5bs70q009xq2x00000gn/T/arduino_build_643615/thermo_270.ino.elf" "/var/folders/w2/jb9h2v3j5s5bs70q009xq2x00000gn/T/arduino_build_643615/thermo_270.ino.hex"
Using library SPI at version 1.0 in folder: /private/var/folders/w2/jb9h2v3j5s5bs70q009xq2x00000gn/T/AppTranslocation/AA37A472-D2D8-4297-9EF6-726E8537F2C2/d/Arduino-2.app/Contents/Java/hardware/arduino/avr/libraries/SPI 
Using library EPD_FLASH in folder: /Users/shirishbytelens/Documents/Arduino/libraries/EPD_FLASH (legacy)
Using library EPD_V231_G2 in folder: /Users/shirishbytelens/Documents/Arduino/libraries/EPD_V231_G2 (legacy)
Using library EPD_PANELS in folder: /Users/shirishbytelens/Documents/Arduino/libraries/EPD_PANELS (legacy)
Using library S5813A in folder: /Users/shirishbytelens/Documents/Arduino/libraries/S5813A (legacy)
Using library EPD_PINOUT in folder: /Users/shirishbytelens/Documents/Arduino/libraries/EPD_PINOUT (legacy)
Using library Adafruit_GFX_Library at version 1.1.5 in folder: /Users/shirishbytelens/Documents/Arduino/libraries/Adafruit_GFX_Library 
Using library EPD_GFX in folder: /Users/shirishbytelens/Documents/Arduino/libraries/EPD_GFX (legacy)
Sketch uses 15382 bytes (6%) of program storage space. Maximum is 253952 bytes.

Global variables use 12138 bytes (148%) of dynamic memory, leaving -3946 bytes for local variables. Maximum is 8192 bytes.
Not enough memory; see http://www.arduino.cc/en/Guide/Troubleshooting#size for tips on reducing your footprint.
Error compiling for board Arduino/Genuino Mega or Mega 2560.`

how do I resolve this?

shirish47, 2017-01-31:
I have commented all lines in program and everything it seems that EPD_GFX G_EPD(EPD, S5813A) is taking this much space itself. 

I really want to use EPD_GFX library for text.

shirish47, 2017-01-31:
ok I found out those buffers are too big in size for MEGA.. is there any way to use EPD with text on mega? or BLE nrf51822 on arduino platform?

hxw, 2017-02-03:
It would take significant code changes, and the addition of an external SPI RAM for the graphics buffer to use this in graphics mode.

The alternative would be to have a character based frame buffer + a fixed width bitmap font and do the lookup as the image is being transferred, this would not use the graphics library but would directly render the bitmap to the panel.  The disadvantage would be that only a single size of font would be possible, it would basically be an emulation of an old style tty display.