repaper/gratis #56: does not work with energia MSP430

shirish47, 2017-02-01:
`demo_270:65: error: #error MSP430: not enough memory
/Users/shirishbytelens/Documents/Energia/libraries/EPD_V231_G2/examples/demo_270/demo_270.ino:93:51: fatal error: Images.h: No such file or directory
compilation terminated.
Using library SPI at version 1.0.0 in folder: /private/var/folders/w2/jb9h2v3j5s5bs70q009xq2x00000gn/T/AppTranslocation/BE951230-8E18-41ED-98C5-FAC968D6B34B/d/Energia.app/Contents/Java/hardware/energia/msp430/libraries/SPI 
Using library EPD_FLASH in folder: /Users/shirishbytelens/Documents/Energia/libraries/EPD_FLASH (legacy)
Using library EPD_V231_G2 in folder: /Users/shirishbytelens/Documents/Energia/libraries/EPD_V231_G2 (legacy)
Using library EPD_PANELS in folder: /Users/shirishbytelens/Documents/Energia/libraries/EPD_PANELS (legacy)
Using library S5813A in folder: /Users/shirishbytelens/Documents/Energia/libraries/S5813A (legacy)
Using library EPD_PINOUT in folder: /Users/shirishbytelens/Documents/Energia/libraries/EPD_PINOUT (legacy)
exit status 1
#error MSP430: not enough memory`

somehow this I am not able to work with paper. some controllers don't have enough memory for 2.7 inch displays.

shirish47, 2017-02-02:
also comparison of what is not working .. please need help...
https://docs.google.com/spreadsheets/d/1IHwUQKCwTqUh-dNvoXu55XYCcxcrQBPDtUeaAa9gsTI/edit?usp=sharing

hxw, 2017-02-03:
Not with the current code as it is using bit mapped graphics and needs enough memory for a full pixel copy of the panel.  I see two possibilities 
- Use a text buffer an modify the code to render the characters on the fly assuming there is enough flash to   hold the font
- use an external SPI RAM to hold the frame buffers; maybe even need SPI FLASH to hold fonts depending on how small a microcontroller you are using