repaper/gratis #57: The upload "u" command in command sketch

LimAnnY, 2017-02-02:
While using the upload command, I am not able to upload any of sketch or xbm file, and i can't even type anything in serial terminal ?why is this happening, and any terminal can be used to upload the xbm file?thank you

hxw, 2017-02-03:
The upload command start a process where the sketch reads ASCII bytes from the serial ports and extracts the hex data from this.  Therefore you need a terminal emulator that supports an ASCII upload of the XBM file.  One problem that can occur is that the upload is too fast and overwhelms the serial input buffer, so it is best if the upload program is able to add a few milliseconds of delay at the end of each line of data.

The upload does not echo anything until it has read the entire image, so if the buffer was overrun and bytes lost then it will be likely just waiting for more input.

I never had any success using the Arduino UI built in serial terminal with this command.  Though you might be able to copy/paste the XBM file a line at a time.  I usually used picocom (BSD/Linux).

LimAnnY, 2017-02-03:
thanks for your recommendation, sorry for asking another question, it mentions the serial terminal with ASCII upload capability ,is there any recommended? I would like to give it a try, the one i am using might be not compatible. thanks

hxw, 2017-02-15:
If you are on Windows, possibly PuTTY has this capability (I do not use Windows so do not really have any other useful suggestions)