repaper/gratis #58: Arduino upload command for epaper

LimAnnY, 2017-02-13:
Hi, i have used the arduino upload command to specific sectors,but then even if the serial monitor shows the read="sth", but when i try to display it out from the specific sectors, it shows empty(blank), as show below:
[image]

Can explain what it actually happens?

hxw, 2017-02-13:
What doe the  "d" command show for the given sector e.g. `d00 8` to show the firs few bytes.  Also `d01 8` to show some bytes from the next sector as the 2.7" display uses 2 adjacent sectors.
I suspect that the data is not getting written to the Flash chip.

Perhaps you could try using the flash_loader program to store an image, then use the command program 'd' command to see if the data is uploaded correctly.  If some data was uploaded, the "i" command should display it as both command and flash_loader use compatible image layout.
(The default setting of flash_loader is to store cat image in sector 0; but it can me modified to store images in other sectors or display pre-loaded image list)

LimAnnY, 2017-02-13:
Hi, I check the d command alrdy, it didnt uploaded to the spi chip on the display board, is there something wrong with the flash library?

hxw, 2017-02-15:
Was the flash_loader able to program the FLASH? What is the 'd' command showing? (All 00 or all FF)

LimAnnY, 2017-02-15:
Hi, flash_loader doesn't load the image, and d command showing all 00

hxw, 2017-02-15:
do other unused sectors also show 00 and how about the erase 'e' command;  does that reset a sector back to all FF?  The fact you get the type code back suggests the FLASH is responding - at leas in a limited way.

LimAnnY, 2017-02-18:
[image]
Hi, for the write class, write enable in flash library, why the this->CS set to LOW instead of high? Supposingly the flash CS must be high right?

LimAnnY, 2017-02-18:
Hi, even the erase function also cannot write the flash into 0xff, but the memory remain 00

hxw, 2017-02-20:
In general CS pins tend to be active low and often have internal pullups (this is often true of other control signals to allow them to be used in a Wired-OR mode = many open-drain outputs connected to a single input)

I just looked at the data sheet for the flash chip (MX25V8005) and there are a couple of extra pins called `hold (pin 7)` and `wp (pin 3)`.
Both pins must be high for normal operation.  In particular if the `wp` pin is low then all writes are ignored.
Could you verify that these two pins are high (3.3V) to rule out PCB problem or that flash chip pin solder broken.

LimAnnY, 2017-03-06:
Hi, i can finally upload the sketch alrdy, now having a problem, when i integrate the arduino with wifi shield and the display is connected to the wifi shield , the flash chip stated is not the correct flash chip, is it because of i stack the wifi shield in between the display and arduino?

hxw, 2017-03-07:
I would suspect the WIFI uses SPI, so is probably using the same CS pin as the flash chip.  so you may have to examine the code to see whether it is easier to change the pin used by the WIFI or the FLASH (probably the FLASH is easier to change).  Also look to see if the WIFI module uses any other pins in common with the FLASH/DISPLAY

LimAnnY, 2017-03-07:
Hi, the wifi shield i am using is cytron wifi shield, here is the module specification
http://www.robotshop.com/media/files/pdf/ESPWiFi-Shield-Rev2-datasheet.pdf
the pin  i use for display is the default one, the only clash is on pin 4 only, even i change the pin position still cannot work

hxw, 2017-03-08:
The pins used by the panel/flash are in: `Sketches/libraries/EPD_PINOUT/EPD_PINOUT_Arduino.h` did yo also make sure WIFI RX/T were ok  (not sure, but D0/D1 might be compatible)

LimAnnY, 2017-03-08:
[image]
what is EPD error=1??what kind of error is that?

hxw, 2017-03-10:
This is failure to detect the E-Ink controller. It suggests something is affecting the SPI data lines or there is a Chip Select conflict.

LimAnnY, 2017-03-13:
Hi, my screen seem like got something wrong, i upload an image onto it, and then i upload another image and display it, the previous image not disappear but overlap  with the present one, is that normal?

LimAnnY, 2017-03-13:
https://scontent.fkul4-1.fna.fbcdn.net/v/t34.0-12/17321771_1464290216928030_950554068_n.jpg?oh=a69c9734fcd9dcd573bea31980c669d4&oe=58C9ACCF

even i put w command toclear the screen, the content still remain,but can see the screen flash for few times

LimAnnY, 2017-03-13:
[image]
even i put w command, the content remains

hxw, 2017-03-14:
Looks like some kind of connection problem - I have had some display like that when running on Raspberry Pi (especially after updating the kernel).  The best way I found to reset/clear the display was to use Arduino(or Ti LaunchPad) running the demo program and let it flip between the two images for a few cycles and even pressing reset occasionally as the first action of the demo is to clear the panel.

LimAnnY, 2017-03-16:
Hi,i have tried to power the arduino with 9v battery through thr power jack,the display get the power supply from3.3v pin,but it cannot works can red led keep flashing,but if I supply the power through usb power bank,it can work?is there any retriction for power input?

hxw, 2017-03-21:
If you are using the small 9V battery then I think it cannot really supply enough current as the the arduino uses linear regulator. You might use 4*AA, possible 3*AA might work alternatives are USB PSU or portable USB battery pack.  There are also some LiPo batteries and charging circuits available at distributors like SparkFun.  There are also those 18650 3.7V rechargables for LED torches, but not sure about charging them 

About USB battery pack, you will have to test it as I have found problems with them:
(I was trying to use one as RaspberryPi "UPS")

- Too low current draw causes auto turn off after a few minutes
- Connecting / disconnecting USB PSU caused long enough power off state to reboot

LimAnnY, 2017-04-10:
Hi, if the serial monitor shows EPD error =2, what kind of error  on EPD is that?

shawaj, 2017-04-10:
Think that error would be panel broken. @tvoverbeek is this correct?

shawaj, 2017-04-10:
Although I'm not 100% sure of that. Does it not print the error message in full?

tvoverbeek, 2017-04-11:
I browsed through the Arduino code on the repaper/gratis github and EPD error is indeed Panel Broken.
Also if you use the incorrect library for your display size, you get very strange effects.