repaper/gratis #61: Error with epd-fuse.service starting - see issue 51 also

francesco-vannini, 2017-03-15:
Re-opening issue #51 as it is still not fixed and does not work correctly.

Me and @shawaj have been testing this. It seems that there is an error in the makefile? It compiles correctly now, but we get the following error when trying to start the service:

```
Job for epd-fuse.service failed. See 'systemctl status epd-fuse.service' and 'journalctl -xn' for details.
```

When checking `systemctl status epd-fuse.service` we get the following errors:

```
● epd-fuse.service - A Fuse driver for the Repaper.org EPD
   Loaded: loaded (/usr/lib/systemd/system/epd-fuse.service; enabled)
   Active: failed (Result: exit-code) since Wed 2017-03-15 18:11:20 UTC; 10s ago
  Process: 3592 ExecStartPre=/usr/bin/mkdir -p ${EPD_MOUNTPOINT} (code=exited, status=203/EXEC)

Mar 15 18:11:20 raspberrypi systemd[1]: epd-fuse.service: control process exited, code=exited status=203
Mar 15 18:11:20 raspberrypi systemd[1]: Failed to start A Fuse driver for the Repaper.org EPD.
Mar 15 18:11:20 raspberrypi systemd[1]: Unit epd-fuse.service entered failed state.
```
Any further thoughts?

francesco-vannini, 2017-03-15:
issue #53 seems to show a similar issue?

francesco-vannini, 2017-03-15:
Just been playing with this a little more.

It seems the issue is due to what is mentioned in #53 - that mkdir on the RasPi is not located by default at /usr/bin/mkdir but is instead located at /bin/mkdir

Addressing that change allows it to install correctly and the epaper screen to work, but then we are stuck not being able to change the screensize. When changing the screen size in the config file located at /etc/default/epd-fuse it then gives the following error:

```
● epd-fuse.service - A Fuse driver for the Repaper.org EPD
   Loaded: loaded (/usr/lib/systemd/system/epd-fuse.service; enabled)
   Active: failed (Result: exit-code) since Wed 2017-03-15 19:05:06 UTC; 3s ago
  Process: 23945 ExecStartPre=/bin/mkdir -p ${EPD_MOUNTPOINT} (code=exited, status=1/FAILURE)
 Main PID: 18417 (code=killed, signal=QUIT)

Mar 15 19:05:06 raspberrypi mkdir[23945]: /bin/mkdir: cannot create directory ‘/dev/epd’: File exists
Mar 15 19:05:06 raspberrypi systemd[1]: epd-fuse.service: control process exited, code=exited status=1
Mar 15 19:05:06 raspberrypi systemd[1]: Failed to start A Fuse driver for the Repaper.org EPD.
Mar 15 19:05:06 raspberrypi systemd[1]: Unit epd-fuse.service entered failed state.
```
Restarting the machine seems to fix this, but surely there is a better way than restarting the Pi each time you want to change screen size?

Additionally, the latest in the gratis folder seem to cause issues with the operation of the epaper screens - permanently leaving lines and other artifacts on the screens despite clearing and removing power etc. Has something been changed in the drivers as well?

francesco-vannini, 2017-03-15:
Perhaps using the command `sudo fudermount -u /dev/epd` would then fix it?

@hxw what is your opinion on the `/usr/bin/mkdir` issue detailed above? and the best way to fix it for Raspbian?

CC @shawaj

shawaj, 2017-03-22:
Closed via #65

closed 2017-03-22 by shawaj