repaper/gratis #66: Do we need a terminating '\n' in the error strings (/dev/epd/error)?

tvoverbeek, 2017-03-23:
In the recent addition of /dev/epd/error in epd_fuse.c the strings representing the error codes (output of 'cat /dev/epd/error') do not have a terminating '\n'. Other strings, e.g. /dev/epd/panel and /dev/epd/version, have a terminating '\n'.
Should a '\n' be added to the error text definitions?

hxw, 2017-03-28:
actual error text definitions should not have any '\n' but the the access to /dev/epd/error should output a '\n' to conform to other usage.

tvoverbeek, 2017-03-28:
The error text definitions are only used for /dev/epd/error output. I'll add the '\n' and submit a new PR.

tvoverbeek, 2017-03-28:
Just submitted a PR (#67) with the '\n' addition to the error texts.

tvoverbeek, 2017-03-30:
PR is merged now. Closing issue

closed 2017-03-30 by tvoverbeek