repaper/gratis #71: Support for 2.87" three color Display (Pervasive; Spectra; E2271CS021)

barthel, 2017-06-05:
I know the limitation of support for displays less than 3".
If there is a chance to support a 2.87" display?

This specific display is a three color (black, white, red) display.
WDYT about supporting three color displays too?

shawaj, 2017-06-05:
We will be supporting the new iTC and Spectra displays later this year :-)

shawaj, 2017-06-05:
FYI, this includes the larger ones... Up to 12"

barthel, 2017-06-05:
Awesome!
Is there a roadmap?
So this work includes support requested by #13 and #45?

shawaj, 2017-06-05:
No official roadmap right now. We're a small team :-)

The 10.2 and 4.41 inch displays use the eTC so they won't ever feature until Pervasive and eInk open source the waveforms for those displays as they're currently completely closed.

The 2.87", 4.2", 7.4" and 12" I believe are the new iTC displays which feature an internal timing circuit in the driver chip and those are available soon. The Spectra displays are the same and have the iTC chip

repaper, 2017-06-06:
2.87" & 4.2" Aurora (BW, black/white) and Spectra (BWR, black/white/red) are launched on [PDi website](http://www.pervasivedisplays.com/products#iTC) a couple of months ago. Yes, they are iTC.
No more displays with eTC driver embedded.  The new 7.4" and 10.2" are with new iTC driver hence you don't need a Tcon board in between. Developer can focus on the interface commands without dealing with the waveform driving.