repaper/gratis #72: Arduino UNO EG020BS011 EPD Extension Board Gen2

macvenez, 2017-07-05:
Hi.
Today I tried this [display](https://www.digikey.it/product-detail/it/pervasive-displays/EG020BS011/EG020BS011-ND/4898784) connected to [EPD Extension Board Gen2](http://www.pervasivedisplays.com/kits/ext2_kit) and my Arduino UNO.
I followed [this guide](https://learn.adafruit.com/repaper-eink-development-board/wiring-the-uno-and-leonardo) and switched the DIP switch for the 2.0" eTC display so 0000001.
I loaded the demo.ino sketch and set the screen size and type (EPD_V230_G2, SCREEN_SIZE 200).
I uploaded the sketch without any errors to my UNO.
I opened the Serial Monitor and the program was running but on the EPD display nothing happened (always white, like it was when I put it out of the box).
Sometimes I read "EPD: DC failed" or "EPD: ok" but nothing happens to the display.
Furthermore as temperature value I get strange values: "Temperature = -537 Celsius".
I think my wiring is ok as I get "EPD FLASH chip detected OK" in the serial monitor but I'm not sure about it so if someone could give me instructions about the wiring for Arduino UNO I would be happy.
I don't know where could be the problem, I attach a screenshot of my serial monitor.
Thank you very much.
 
[image]

Hardy-PDi, 2017-07-11:
Hello Macvenez,
You are going to have a level shifter to bridge the control interface between Arduino UNO and EXT2 board.
Our EXT2 board is not available for 5V system. You have to add 6 control signals for 3V below,
- SCL_PIN   //EXT2 BOARD J5 pin 7
- SDA_PIN   //EXT2 BOARD J5 pin 15
- CS_PIN    //EXT2 BOARD J5 pin 19
- DC_PIN    //EXT2 BOARD J5 pin 9
- RESET_PIN  //EXT2 BOARD J5 pin 10
- BUSY_PIN   //EXT2 BOARD J5 pin 8
Don't forget to pull high Panel ON(J5 pin 11). This pin can operate at 5V

macvenez, 2017-07-11:
Thank you for the answer.
I'll try it as soon as possible.
In the meantime I tried the board with raspberry and it worked perfectly
immediately.
I discovered there's no reference for the commands available for pi... any
solution?
Thank you very much