repaper/gratis #78: your homepage is broken

gauteh, 2017-09-16:
not possible to access documentation and reference links

shawaj, 2017-09-16:
Thanks, will look into this

Is there specific documents you are looking for?

garylzimmer, 2018-02-07:
Any update on you having a functioning website?

shawaj, 2018-02-07:
What are you looking for?

techydude, 2018-05-03:
I emailed you about this last week.
specifically:
https://repaper.org/doc/cog_driving.html
https://repaper.org/doc/em027as012.html
https://repaper.org/doc/extension_board.html
https://repaper.org/doc/downloads/epd_ref_circuit_v05.pdf

molecule, 2018-07-09:
Any update on this? All the docs I need are giving 404 Page Not Found Errors.

Specifically:
1. Click on this link: https://www.repaper.org/reference/
2. Open the "datasheets" tab below the video.
3. Try to open any of the listed documents.

joedeveloper, 2018-08-27:
It is unfortunate that this is still an open and serious issue with the website.
It gives me pause as a person researching options for epaper libs and hardware suppliers.

shawaj, 2018-08-27:
All of the documents are on the pervasive displays website

joedeveloper, 2018-08-27:
Then link there :D start in this thread

shawaj, 2018-08-27:
http://www.pervasivedisplays.com/

molecule, 2018-08-30:
Really? A link to the homepage? You could at least link more directly:

1.44": http://www.pervasivedisplays.com/LiteratureRetrieve.aspx?ID=238015
2.0": http://www.pervasivedisplays.com/LiteratureRetrieve.aspx?ID=238016
2.71": http://www.pervasivedisplays.com/LiteratureRetrieve.aspx?ID=238025

But part of the issue is that Pervasive Displays sometimes removes links, so I was hoping you would host them separately and help avoid lost documentation pdfs in the future. Also their site is not as easy to navigate as yours, so it would have been nice to see them linked by you. Seems like just as much effort to add these link to your site as to reply here.